system76/ec #384: flash_internal: Err(Protocol(1))

urv, 2023-07-23:
Hi
I have a system76/oryp9 and when running flash_internal I get Protocol(1) error. 
Does anyone encountered the same issue and knows how to resolve this? Is it related to system76_ectool warning?

Here is the complete output:
```
make BOARD=system76/oryp9 flash_internal     
warning: unused manifest key: registries
warning: use of deprecated unit variant `hidapi::HidError::OpenHidDeviceError`
   --> src/main.rs:430:43
    |
430 |                     Err(hidapi::HidError::OpenHidDeviceError.into())
    |                                           ^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(deprecated)]` on by default

warning: `system76_ectool` (bin "system76_ectool") generated 1 warning
    Finished release [optimized] target(s) in 0.00s
file board: Ok("system76/oryp9")
file version: Ok("2023-06-08_485f390")
ec board: Ok("system76/oryp9")
ec version: Ok("2023-06-08_36c78ea")
Waiting 5 seconds for all keys to be released
Sync
Result: Err(Protocol(1))
Sync
System will shut off in 5 seconds
Sync
failed to flash 'build/system76/oryp9/2023-06-08_485f390/ec.rom': Protocol(1)
make: *** [src/board/system76/common/common.mk:117: flash_internal] Error 1
```

Cheers

jackpot51, 2023-07-23:
On new systems there is firmware locking that must be unlocked prior to flashing. I recommend using the flash script from the firmware-open repository which handled this automatically

urv, 2023-07-23:
Thanks a lot Jeremy - that was indeed the issue; with the script I could flash.

closed 2023-07-23 by urv