system76/ec #390: Sync fans based on temp instead of duty

crawfxrd, 2023-07-29:
The current implementation syncs all fans to the highest duty.

Replace this with syncing to the highest temperature reported by the sensors.

This will allow independent curves for FAN0 and FAN1 that work as expected based on their actual points.

qnxdev, 2023-10-16:
I highly support this, my darp9 is super hot sometimes, I have to turn on fan manually with `Fn + 1` but why can't it do that  job?

When I use fan it cools down the laptop such that I can put on my lap & work.

leviport, 2023-10-16:
@qnxdev this is unrelated to what you're describing. The darp9 only has a single fan, and this issue is about fan speeds on models with dual fans.

closed 2024-08-15 by jackpot51

swisscdn, 2024-08-16:
> @qnxdev this is unrelated to what you're describing. The darp9 only has a single fan, and this issue is about fan speeds on models with dual fans.

So shall I raise another issue for this?

leviport, 2024-08-16:
@swisscdn No need. User-customizable fan curves are already in the works.

PR #477 (2024-08-15): Sync fans based on temp instead of duty
Fans may be different sizes, placed in asymmetrical positions, or have different amounts of venting through the chassis. These characteristics affect the ability for each fan, separately, to dissipate heat and generate noise.

Replace syncing fans to the highest duty calculated for each fan, based on separate thermal sensors, to using the highest reported temperature across all sensors to calculate each fan's duty for that highest temperature.

In other words: The old behavior synced fans based on the *output* value (duty), while this new behavior syncs  [... cut]