system76/ec #407: UEFI update application appears to possibly brick the EC every time

ilikenwf, 2023-09-11:
I've been having to update it via Arduino...

I have tried both with secureboot enabled and disabled, however once the flash process completes and the machine is supposed to power off/reboot, nothing happens and the power button is unresponsive. Do I just need to wait longer? When I flash with an arduino it comes right on...and to that point the updater launches and recognizes the new, updated EC flash matches the version it has available.

crawfxrd, 2023-09-11:
- https://github.com/system76/firmware-update/blob/b98de656b212964633f7875bb6b172fe20732a6b/src/app/ec.rs#L801-L807

After flashing it waits 5 seconds then triggers a WDT reset.

ilikenwf, 2023-09-11:
I gave it longer than 5 seconds...I don't even get a power button LED. I know you QA everything very hard with these updates, so if it is something not specific to me I suppose your QA guy will catch it when he tests on a serw13?

crawfxrd, 2023-10-13:
Do you still have this issue with `2023-09-19_16ef69c`, or newer if you are flashing custom images?

ilikenwf, 2023-10-13:
I will have to attempt a new build and flash, when I do I will let you know.

closed 2023-12-22 by crawfxrd