system76/ec #408: Current master appears to prevent Keyboard LED Control Using Fn Keys When Using system76-acpi-dkms module

ilikenwf, 2023-09-11:
After flashing, I notice that I can have the expected behavior if I unload the acpi module, however if I load it, the keyboard LEDs are immediately turned off and may only then be set using the keyboard configurator.

ilikenwf, 2023-09-11:
Of course this could be that the kernel module may need updated to coincide with the new EC firmware when released as well.

crawfxrd, 2023-09-12:
What kernel are you on? You probably need the DKMS module for now.

ilikenwf, 2023-09-12:
i usually run zen, but it has the same version as the DKMS repo. If I load the module up, though, the KB lights go out...

Right now I'm just using the configurator to set the LEDs how I like them. This does happen both with and without the draft to not reset the LEDs on wake - but with a build that has that patch, the keys light back up as I had them when I open the lid and wake it back up, at least.

ilikenwf, 2023-09-12:
This may be an issue with Arch and other distros? I notice I also have no num/caps lock indication.

https://old.reddit.com/r/System76/comments/15j4b74/oryx_pro_10_arch_keyboard_backlights_nonfunctional/

crawfxrd, 2023-09-12:
> If I load the module up, though, the KB lights go out...

That's expected: The module will reset the LEDs when loaded.

> I notice I also have no num/caps lock indication.

There aren't any.

ilikenwf, 2023-09-12:
Good to know; however loading the module, the reset persists unless I use software controls for the LED lights...the hardware/FN keys no longer control the keyboard LEDs.

Are the keyboards in these the individually addressable LED keyboards that your manufacturer sometimes uses, and just need further devleopment to expose that functionality, or are they different? If they're individually addressable with later development, that would allow for num/caps to have different colors when active.

crawfxrd, 2023-09-12:
serw13 has an IT8295FN LED Controller for per-key RGB control. It is exposed to the host via USB port 6, and I guess can communicate with the EC over SMBus (`SMD_VGA_THERM`). However, this is outside the scope of the issue.

ilikenwf, 2023-09-12:
True - thanks for the info.

It does seem this may be a "non Pop distros" issue or something specific to arch. I'll try with the stock arch kernel later but do know that the DKMS stuff is built for Zen, though it brings it's own in tree version of the acpi driver.

ilikenwf, 2023-09-12:
I think when I got my serw13 that it came with an ec and firmware build from May? Remind me the commit hash and I can go back to that and test again...or git bisect if you really want me to...or something.

crawfxrd, 2023-09-13:
The current release of serw13 is system76/firmware-open@9560b2e.

ilikenwf, 2023-09-13:
I'm going to close this one; I discovered that the PKGBUILD I was using for whatever reason wasn't using git to build and was out of date. The updated version fixes the hardware keys for me as well.

For the individual addressing of LED keys, @crawfxrd - I'm not sure I can actually see this device from userspace, is it something that would need further exposing through the EC/firmware in order to be addressable? In theory, it should be possible to get working with openrgb if it gets more exposed/documented.

Even if not, devices with individual key RGB could have num lock and caps lock indicators added at the EC level, perhaps?

closed 2023-09-13 by ilikenwf

leviport, 2023-09-13:
I believe only bonw14 and bonw15 had per-key RGB. The keyboard on serw13 only has the [small backlight cable](https://tech-docs.system76.com/models/serw13/repairs.html#replacing-the-keyboard) which is typical for all of the single-zone backlit keyboards. The [backlight connectors on bonw15](https://tech-docs.system76.com/models/bonw15/repairs.html#replacing-the-keyboard) are much more complicated.

crawfxrd, 2023-09-13:
Ah right. "Full Color "Per Key" LED" is an ODM option which is why the schematics show the IT8295FN. We have "Multi Full Color LED Keyboard" which just has RGB zones.

ilikenwf, 2023-09-13:
Makes sense, thank you for the clarification.