system76/ec #412: Keyboard unresponsive after `lemp10` firmware update

AaronFeickert, 2023-10-09:
- Model: `lemp10`
- BIOS version: `2023-09-08_42bf7a6`
- EC version: n/a
- OS: `Pop!_OS 22.04 LTS`
- Kernel: `6.5.4-76060504-generic`

After the latest firmware update, leaving the laptop locked for several minutes (presumably entering some kind of suspend state) leaves the keyboard permanently unresponsive and the trackpad/mouse sometimes briefly unresponsive.

### Steps to reproduce

Lock the screen and wait for the display to turn off. Press a key to activate the display and return to the unlock prompt. The keyboard and trackpad work as expected.

Lock the screen and walk away for a while. (The laptop presumably enters a suspend state, despite this not being specified in the power settings.) Press a key to activate the display and return to the unlock prompt. The keyboard is permanently unresponsive. The trackpad is sometimes briefly unresponsive. The only way to restore functionality is a hard reset via the power button.

If Bluetooth is on and a keyboard is connected, the Bluetooth keyboard works as expected.

### Expected behavior

The keyboard and trackpad should accept input.

### Actual behavior

The built-in keyboard does not accept input (but does function to activate the display). The trackpad sometimes briefly does not accept input. If a Bluetooth keyboard is connected, it accepts input.

### Additional info

This behavior only began after a firmware update.

crawfxrd, 2023-10-10:
Also happens on b967e7c92158 + b63e2092ceca, so definitely seems related to eSPI rather than my KBC/KBScan changes.

I have also not seen it happen after suspending via systemctl, so it also seems to be specifically opportunistic suspend.

crawfxrd, 2023-10-10:
I'm also not seeing it if I switch to the PECI-over-eSPI implementation, which would mean it's another PECI/eSPI issue similar to system76/firmware-open#424.

crawfxrd, 2023-10-10:
I suspect the lockup is caused the the infinite waits in the legacy PECI implementation.

closed 2023-10-12 by crawfxrd

crawfxrd, 2023-10-13:
2023-10-13_5d997d1 staged for release.

AaronFeickert, 2023-10-17:
Tested `2023-10-13_5d997d1` on my `lemp10`, and the issue appears to be solved! Thanks to @crawfxrd and team for the prompt action on this.

PR #414 (2023-10-12): peci: Add timeouts to infinitely blocking waits
Add timeouts to the legacy PECI implementation to prevent the EC locking up when PECI stops working, such as during S0ix opportunistic suspend.

This is not the optimal solution, as PECI should not be available at all to cause the lock up in the first place, but it at least prevents the issue.

Test:

- [ ] PECI works as expected during S0 working state (fans not stuck at 50%)
- [ ] System no longer locks up after screen blank on TGL

Resolves: #412
Ref: #389