system76/ec #424: `make flash_internal` fails

leviport, 2023-12-29:
When testing #423, I noticed that `flash_internal` wasn't working on a serw13. After checking `master` back out, I'm still saying the same behavior (`BOARD=system76/serw13` was specified in a config.mk in this case and builds successfully without `flash_internal`):

```
system76@serw13-2k:~/firmware-open/ec$ make flash_internal
    Finished release [optimized] target(s) in 0.08s
[sudo] password for system76:
file board: Ok("system76/serw13")
file version: Ok("2023-12-12_6f1e653-dirty")
ec board: Ok("system76/serw13")
ec version: Ok("2023-12-12_6f1e653-dirty")
Waiting 5 seconds for all keys to be released
Sync
Result: Err(Protocol(1))
Sync
System will shut off in 5 seconds
Sync
failed to flash 'build/ec.rom': Protocol(1)
make: *** [src/board/system76/common/common.mk:136: flash_internal] Error 1
```

The machine does not shut off after this.

crawfxrd, 2023-12-29:
Documentation needs to be updated to clarify internal flashing will not work if firmware is locked (`CONFIG_SECURITY=y`).

leviport, 2023-12-29:
Ah yep, that explains it. I wonder how many times I've forgotten that.

PR #427 (2024-01-03): docs: Add note about write lock for flashing
Fixes: #424
closed 2024-01-03 by jackpot51