system76/ec #447: Feature Request/Suggestion: Remember Camera Toggle State or Make it Configurable in Settings

ilikenwf, 2024-03-06:
I rarely if ever use the camera, it would be nice to have the ability to have the EC either use the last used state (enabled/disabled) or allow me to set a default of enabled/disabled in the settings.

I'm using a serw13, however I'm sure it would be welcome in all models with the feature.

crawfxrd, 2024-03-06:
Camera power is always enabled. It's not configurable.

ilikenwf, 2024-03-06:
Then why does the camera in the serw13 disappear from lsusb when the disable hotkey is used?

crawfxrd, 2024-03-06:
Oh, I didn't even realize that that's what we used that icon for.

Anyways, the only persistent "config" is the custom keyboard layout. Any other settings, like battery charging thresholds, are lost on EC reset.

See: #256

ilikenwf, 2024-03-06:
My current custom build includes some of the pending pr's for backlight settings on the keyboard, but yes, it's a handy feature, though with the open EC, mostly open UEFI, and in my build the ME fully stripped, there's little to no actual threat at the low level...

Everything threatening is in userspace.