system76/ec #513: Use temperature-based hyteresis

crawfxrd, 2025-01-04:
We currently use a time-based hysteresis of the duty (output value). This method results in poor behavior of fans.

With `HEATUP`, it can result in:

- slow response of increasing temps that could lead to the CPU throttling before fan reaches its max point

With `COOLDOWN`, it can result in:

- fans being held at higher duties than necessary over periods of time
  - such as short temp spike at start of a CPU-intense workload like compilation
- not reducing temps sufficiently below the point threshold, resulting in rapid oscillation of fan duty
  - noticeable at point 0, which can result in near-constant fan on/off

Use a temperature-based (input value) hysteresis instead.

earlruby, 2025-02-03:
I was just looking at the source code trying to figure out how to do this. Are you actively working on this problem or just noting the issue hoping someone will pick it up?

Any idea why [the fan duty cycle is set to never exceed 65%, no matter how hot the system gets?](https://github.com/system76/ec/blob/13dd6a10381e3973006fd68958a7b57c2068b488/src/board/system76/common/peci.c#L48-L49)

crawfxrd, 2025-02-03:
Initial attempt done in [`fan-temp-hyst`](https://github.com/system76/ec/commits/fan-temp-hyst/).

earlruby, 2025-02-03:
Nice!