system76/ec #525: Zero temperature reading from ACPI

nextsux, 2025-03-31:
Since I've started to use new USB-C Phillips monitor, I've noticed fans turned on almost all the time. After some investigation I've noticed pretty unusual readings from ACPI:

```
$ sudo sensors                                            
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +49.0°C  (high = +110.0°C, crit = +110.0°C)
Core 0:        +47.0°C  (high = +110.0°C, crit = +110.0°C)
Core 1:        +47.0°C  (high = +110.0°C, crit = +110.0°C)
Core 2:        +47.0°C  (high = +110.0°C, crit = +110.0°C)
Core 3:        +47.0°C  (high = +110.0°C, crit = +110.0°C)
Core 4:        +49.0°C  (high = +110.0°C, crit = +110.0°C)
Core 5:        +49.0°C  (high = +110.0°C, crit = +110.0°C)
Core 6:        +49.0°C  (high = +110.0°C, crit = +110.0°C)
Core 7:        +49.0°C  (high = +110.0°C, crit = +110.0°C)
Core 8:        +47.0°C  (high = +110.0°C, crit = +110.0°C)
Core 12:       +54.0°C  (high = +110.0°C, crit = +110.0°C)
Core 32:       +46.0°C  (high = +110.0°C, crit = +110.0°C)
Core 33:       +46.0°C  (high = +110.0°C, crit = +110.0°C)

nvme-pci-0100
Adapter: PCI adapter
Composite:    +30.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +30.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +33.9°C  (low  = -273.1°C, high = +65261.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +36.0°C  

system76_acpi-acpi-0
Adapter: ACPI interface
CPU fan:     9540 RPM
**CPU temp:      +0.0°C**  <<<<<< 0 C here

BAT0-acpi-0
Adapter: ACPI interface
in0:          17.11 V  
curr1:         0.00 A  
```

Where ACPI reports 0C CPU temp. It seems rebooting system/EC fixes the issue for some time but pretty unsure how to continue investigation. What I've found/tried so far:

  - upgrading firmware/ec to latest master of https://github.com/system76/firmware-open.
  - temp works fine after removing battery for some time
  - after connecting/disconnecting Philips display via USB-C temp reading shows 0 for few seconds before it fixes itself.
  - after some time being connected to that USB-C (lid closed) it just stops reading temperature and shows 0. Fans start to blow.

Experiencing this on `lemp13-b`

If you have any ideas on how to debug/fix this issue. I'm more than happy to help.

nextsux, 2025-03-31:
Is there anything I can try? The issue unfortunately seems to be there for pretty long time without any workaround :(

crawfxrd, 2025-03-31:
Running `make console_internal` and then reproducing the issue will generate logs to see if it's the PECI timeout issue.

nextsux, 2025-03-31:
at this point I'm getting just spam of:

```
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
```

I'll get it rebooted and log to the file to get the moment when it starts

nextsux, 2025-04-01:
longer log:

```
Battery:
  Temperature: 0B9F
  Voltage: 42D5
  Current: FBF7
  Charge: 0064
  Status: 00E0
Charger (oz26786):
  ChargeCurrent: 0000
  ChargeVoltage: 0000
  ChargeOption_1: 0802
  ChargeOption_2: 0800
  AdapterCurrent: 0000
peci_wr_pkg_config: wait upstream
power_peci_limit 45 = 64
Power adapter unplugged
Battery:
  Temperature: 0BB7
  Voltage: 4424
  Current: 0000
  Charge: 0064
  Status: 00E0
Charger (oz26786):
  ChargeCurrent: 0000
  ChargeVoltage: 0000
  ChargeOption_1: 0802
  ChargeOption_2: 0800
  AdapterCurrent: 0000
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_wr_pkg_config: wait upstream
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: response timeout
power_peci_limit 45 = 0
Power adapter plugged in
peci_wr_pkg_config: wait upstream
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: wait response
peci_wr_pkg_config: response timeout
power_peci_limit 65 = 0
Battery:
  Temperature: 0BB7
  Voltage: 435C
  Current: FDF3
  Charge: 0064
  Status: 00E0
Charger (oz26786):
  ChargeCurrent: 0000
  ChargeVoltage: 0000
  ChargeOption_1: 0802
  ChargeOption_2: 0800
  AdapterCurrent: 0000
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout
peci_get_temp: response timeout

```

what happened here is - connecting to Philips monitor via USB-C, then monitor went to sleep (like after some time of inactivity). Then I've just moved the mouse to wake up the monitor and ended with `peci_get_temp: response timeout` repeating endlessly

crawfxrd, 2025-04-02:
No. Disabling it requires a dedicated EC pin, normally called `H_PECI` in the schematics, connected to the PECI pin of the CPU. lemp13 does not have it connected.

nextsux, 2025-04-03:
Thank you for the explanation 👍

As far as I've read ectools documentation, resetting EC leads to computer reset. Is there any other way to recover from this state? Like before this is fixed having a script that "soft resets" EC or something like that.

In current state I have like 4 options:
  - change notebook 
  - change monitor and hope it behaves differently 
  - disable screensaver / keep the screen always on
  - reboot everytime screen goes to sleep

And all of them are a little bit - "suboptimal" 😅

crawfxrd, 2025-04-10:
Will track in #369.